Jomelo/LCDMenuLib #5: Version V2 beta 2 wire.h issue

pierrotm777, 2015-05-19:
With the version V2 beta 2, the example LCDML_example_1_easy.ino return me always 

C:\tmp\Arduino Compilateur\arduino-1.6.4\libraries\New_LiquidCrystal_V1.3.0\I2CIO.cpp:35:26: fatal error: ../Wire/Wire.h: No such file or directory
 #include <../Wire/Wire.h>

I use the last New_LiquidCrystal_V1.3.0

Jomelo, 2015-05-19:
the new liquid crystal lib needs the wire lib, but when you use the orginal lib it is not used. When you use an extern lib, you have to include in every example libs. The wire lib needs many memory. When you have more questions why wire is needed without i2c then ask the autor of the extern lib. I cannot answer that.

pierrotm777, 2015-05-19:
Thanks , i understand !

closed 2016-01-06 by Jomelo