Jomelo/LCDMenuLib #7: I2C alterable address

pierrotm777, 2015-05-28:
Hello,
Actually, we define the I2C address with the _LCDML_DISP_addr value .
Is it possible to have an fonction that change this value ?
My goal is to read the address from the EEPROM !!!

Thanks for your job !!!

Jomelo, 2015-05-29:
At the moment it is not possible. 
In the version v2.1 i will change the class concept to remove the settings in  ___config.h.
But at the moment i renovate my new appartment and have no time for that.

pierrotm777, 2015-05-29:
Ok, good luck

Jomelo, 2016-01-06:
On current master version it is possible, because i remove the ___config.h and _class.h 
The LCD Object and all LCD function can be set direct in code. 

Now you can use Graphic Displays with this Lib

closed 2016-01-06 by Jomelo