Jomelo/LCDMenuLib #8: support for Adafruit LiquidCrystal Lib

mscosti, 2015-10-19:
I'd like to request that support for the adafruit LiquidCrystal library be added.  It is like the new Liquid Crystal Lib, but is made specifically to support  a MCP23008 I2C backpack (located here https://www.adafruit.com/product/292)

Their library is available for reference here: 
https://github.com/adafruit/LiquidCrystal

Do you think this will be possible? 
Thanks!

Jomelo, 2015-10-19:
I think it is possible. 
Have you tested the lib without the lcdmenu? The  adafruit lib is very old. (4 years and older) I think, that there are problems with the new arduino versions.

mscosti, 2015-10-19:
It works for me, using latest arduino IDE and the Arduino Uno R3. It is the only Liquid Crystal library that I have been able to successfully get working with my LCD + I2C backback (both purchased from adafruit).

After looking through your library I realized it didn't look too hard to add other liquid crystal lib support, and was able to get it working with the adafruit library for I2C.  I have yet to test the other communications the adafruit library supports, however i'd be surprised if there were any problems.

If I have time tonight I'll submit a pull request with my addition? :)

Jomelo, 2015-10-19:
I can add it tomorrow evening. 

The changes are only in LCDMenuLib_class.h  and LCDMenuLib___config.h (Do not forgett to include the lib in *_class.h and in the your code).

mscosti, 2015-10-20:
I already had this basically done, and was able to test it is working, so I submitted a Pull Request with the adafruit addition, so you can review it at your leisure. (I hope you don't mind)

I also noticed an issue with the DogLcd support when I was making the addition, and included a fix for that as well. 

In depth Descriptions are included in the pull request in at #9

mscosti, 2015-10-20:
closed with #9

closed 2015-10-20 by mscosti