Jomelo/LCDMenuLib #13: LCD compatible

StafWars, 2016-03-18:
Dear

I'm using your library with a 3x16 lcd, but is it possible to use it with a dot-matrix lcd?
For example this lcd:
http://www.mouser.be/ProductDetail/Newhaven-Display/NHD-C12864A1Z-FSW-FBW-HTT/?qs=sGAEpiMZZMvkC18yXH9iIjqe6IqxYXYlNQS2QNbO9KE%3d

If so, how do I have to initialize this?

thanks in advance
Greetings

Jomelo, 2016-03-28:
Yes it is possible. 
You have to include your graphic lib in main tab, then change the output format in "LCDML_DISP" tab with the new commands from your graphic lib. 
I haven`t testet it yet but when i have a graphic display i add an example.

StafWars, 2016-04-12:
I've bought an dogm128 glcd, and i'm using u8glib as graphic lib, but can't seem to find the correct notations for the output format. Do you know which commands that I should have to change? I think that i should multiply the setCursor (from the glib) with the font-height. But don't know how to initialize the scrollbar and row's etc.. 
Is there anyone that has already managed to do so?
Big thanks in advance
greetings

Jomelo, 2016-04-12:
Is the output format this one?  
https://github.com/olikraus/u8glib/wiki/device#st7565-128x64

The rows are calculated:
#define _LCDML_u8g_lcd_h       64
#define _LCDML_u8g_font_h      13

rows = _LCDML_u8g_lcd_h / _LCDML_u8g_font_h

When you use a other font like u8g_font_5x10 or other you have to change font_h to 5 ...

When you use more then four rows you have to change the define in LCDMenuLib.h  from 4 to 6 or bigger.

In the newest version an example is included.

StafWars, 2016-04-13:
Yes, that's my output format.
I will try out your example tomorrow!
Many Thanks!

Jomelo, 2016-04-22:
Can i close this issue ?

StafWars, 2016-04-29:
Yes, it can be!
Thanks for the help!

closed 2016-04-29 by Jomelo