Jomelo/LCDMenuLib #14: Encoder Esc-button

StafWars, 2016-03-21:
Hi, me again,
I'm fairly new to arduino, and would like to add an escape-button to the encoder-control when double-clicking the encoder. How could i implement that in the LCDML_control doc?

dragoblaztr, 2017-04-28:
You can add this feature by ussing OneButton Library (it also works setting up on joystick control)

[https://github.com/mathertel/OneButton](url)

Also you must to include it on header of main arduino ino 

` #include <OneButton.h>`

Besides on the header of LCDML_CONTROL over encoder control setup after the definition pins
you should add this lines:

```
  OneButton button(_LCDML_CONTROL_digitalread, true);
  void enter(){LCDML_BUTTON_enter();}
  void quit(){LCDML_BUTTON_quit();}
```

comment this lines

```
// global defines
  uint8_t  g_LCDML_CONTROL_encoder_t_prev = 0;
```

on LCDML_CONTROL_loop

comment the following lines

```
  unsigned char t = digitalRead(_LCDML_CONTROL_encoder_pin_button);

  // change button status if high and low active are switched
  if (_LCDML_CONTROL_encoder_high_active == 1) {
    t != t;
  }
```
and this

`    g_LCDML_CONTROL_encoder_t_prev = 1;`

also this

```
else {
    // check button press time for enter
    if((millis() - g_LCDML_DISP_press_time) >= _LCDML_DISP_cfg_button_press_time) {
      g_LCDML_DISP_press_time = millis(); // reset button press time
      
      // press button once
      if (!t && g_LCDML_CONTROL_encoder_t_prev == 0) {          
          LCDML_BUTTON_enter();          
      } 
      else {
        g_LCDML_CONTROL_encoder_t_prev = 0;
      }
    }      
  }

```

and at last add this code

on LCDML_CONTROL_setup

```
  button.attachClick(enter);
  button.attachDoubleClick(quit);
```

on LCDML_CONTROL_loop before 

`// read encoder status `

add this

`	button.tick();`

You're done

Regards

sgryb, 2017-10-16:
Great, thank you! Great match: OneButton is exactly the lib I am using for my projects :-) 
Menu works fine! Just one correction to your code:
instead of:
   OneButton button(_LCDML_CONTROL_digitalread, true);
should be:
   OneButton button(_LCDML_CONTROL_encoder_pin_button, true);

dragoblaztr, 2017-10-16:
That it's true I forgot to make that correction because actually I made that fix for joystick control, anyway it works for both making that correction, now you can close this issue.

Regards

closed 2017-10-16 by Jomelo

sgryb, 2017-10-16:
I would also try to use an other Rotary Encoder handler based on pin change interrupt... there is wonderful library by Matthias Hertel, http://www.mathertel.de; available @ https://github.com/mathertel/RotaryEncoder

sgryb, 2017-10-16:
OK, just did it! works fine. 
Just a quick question:  having encoder handler based on pin change interrupt, do I still need the following line in the    LCDML_CONTROL_setup    code:

```
  // set encoder update intervall time 
  LCDML_BACK_dynamic_setLoopTime(LCDML_BACKEND_control, 5UL);  // 5ms 
```

dragoblaztr, 2017-10-16:
@sgryb 
You're right. if you use that library you can delete that line, but in LCDML_CONTROL you need to redefine the buttons up and down, it'll not work properly as is.

Can you share your button up / down definitions for who wants to implement this?

Regards

sgryb, 2017-10-16:
Sure, here we go:
1.  include library  to the main arduino.ino code:
`#include <RotaryEncoder.h>`

2. now with both OneButton and RotaryEncoder the entire CONTROL WITH ENCODER section looks like:
`
// *********************************************************************
// *************** (3) CONTROL WITH ENCODER ****************************
// *********************************************************************
#elif(_LCDML_CONTROL_cfg == 3)
// settings
  #define _LCDML_CONTROL_encoder_pin_a           10 // pin encoder b
  #define _LCDML_CONTROL_encoder_pin_b           8  // pin encoder a
  #define _LCDML_CONTROL_encoder_pin_button      A3 // pin taster
  #define _LCDML_CONTROL_encoder_high_active     0  // (0 = low active (pullup), 1 = high active (pulldown)) button
 
  // source  https://github.com/Jomelo/LCDMenuLib/issues/14   
  // source: https://github.com/mathertel/OneButton                                                   
  OneButton button(_LCDML_CONTROL_encoder_pin_button, true);
  void enter(){LCDML_BUTTON_enter();}
  void quit(){LCDML_BUTTON_quit();}       

  // Set global vars for a RoraryEncoder 
  static int16_t enLast = 0;
  int16_t enValue;
  int enSteps = 0;
  boolean up = false;
  boolean down = false;
  // for RotaryEncoder
                                   
  // source: https://github.com/mathertel/RotaryEncoder
  RotaryEncoder encoder( _LCDML_CONTROL_encoder_pin_a, _LCDML_CONTROL_encoder_pin_b );

                                                      
// The Interrupt Service Routine for Pin Change Interrupt 0
// This routine will only be called on any signal change on D8 and D10: exactly where we need to check.
ISR(PCINT0_vect) {
  encoder.tick(); // just call tick() to check the state.
}              

// *********************************************************************
// setup
void LCDML_CONTROL_setup()
{
  // init pins  ->  do not need that; it is done in encoder initiation: RotaryEncoder encoder();
  //pinMode(_LCDML_CONTROL_encoder_pin_a      , INPUT_PULLUP);
  //pinMode(_LCDML_CONTROL_encoder_pin_b      , INPUT_PULLUP);
  
  button.attachClick( enter );
  button.attachDoubleClick( quit );

// You may have to modify the next 2 lines if using other pins than 8-PB0 and 10-PB2 - PORTB
  uint8_t *pcmask=digitalPinToPCMSK(8);
  PCICR |= (1 << digitalPinToPCICRbit(8) );  
  *pcmask |= (1 << digitalPinToPCMSKbit(8)) | (1 << digitalPinToPCMSKbit(10) ); 
}
// *********************************************************************
// loop
void LCDML_CONTROL_loop()
{    
  // check buttons
  button.tick();

  // check Encoder
  enValue = encoder.getPosition();
  enSteps = enValue - enLast;     // I use enSteps for counting encoder clicks for catching fast rotation in main loop()
  if( enValue > enLast ) {
    down = true;
    LCDML_BUTTON_down();
  } else if( enValue < enLast ) {
    up = true;
    LCDML_BUTTON_up(); 
  }
  enLast = enValue;
}
// *********************************************************************
// ******************************* END *********************************
// *********************************************************************
`

fietstasss, 2018-01-15:
How can the above code be adjusted to work with a Arduino Mega 2560 ?  
I've got no reaction on my rotary encoder, think its because of some port mapping problems. Anyone an idea on how to fix this?

Thanks in advance, 
A.

dragoblaztr, 2018-01-19:
Try this

https://www.pjrc.com/teensy/td_libs_Encoder.html

It's about another encoder library so you need to adapt the reading functions of the above code.

Regards

dragoblaztr, 2018-02-16:
@fietstasss 
Could you fix your issue?