Jomelo/LCDMenuLib #15: Empty Receiving Serial Buffer in LCDML_FUNC_DISP

lesikiel, 2016-03-21:
Dear Nils,

I'm currently using your LCDM library and I'm very impressed with your work.

However after quite some programming effort I've run to a brick wall, where I would have to change
everything unless I find a workaroud. Maybe you can help. The issue is as follows:

I'm currently using the multiple hardware Serials of the Arduino to communicate over bluetooth
with another Arduino. Sending my data works just fine, as well as receiving, however the LCDML_FUNC
structure seems to prohibit the emptying of the receiving Serial buffer. This means that with each read
out of a String terminated by '/n' I receive the same string and am thus only able to send one string per
function call. When ending the function and restarting the buffer is emptyed. 

Is there any workaround (hopefully without the implementation of backend functions) for emptying
the Serial buffer inside one LCDML-FUNC call?

Thanks in advance,

Leo

Jomelo, 2017-01-12:
Hi, 
i think this problem comes not from the lib. Only one think can happen, when you are using the serial control with 'a', 'w','d', ... the lib reads the serail buffer. When you use an other method for controling there is no context with this lib.

lesikiel, 2017-01-16:
Hi,
it is true, this must be a Arduino-System error. I had to name the functions SerialEvent() for it to work.
For multiple Serialports you take SerialEvent1(), SerialEvent2(), ... for the respective Serial Ports. I hope this helps someone else, I had to figure that out on my own.

closed 2017-02-08 by Jomelo