Jomelo/LCDMenuLib #16: Debounce encoder with software

StafWars, 2016-05-25:
Hi, 
I would like to debounce the encoder because now I have to be really carefull with the rotary. It doesn't do what I want it to do, and i think it's because it isn't debounced. 
I would like to do the debouncing with software, because i've already made an PCB (maybe too early...)
What would your recommendations be to debounce a rotary encoder on the  LCDML_CONTROL page.
I've worked with delays, but that didn't do much...

Kind regard!

StafWars

Jomelo, 2016-05-25:
The encoder is doubonced over the backend system. The time on "LCDML_BACKEND_control" is to large.

Test following:
LCDML_BACK_new_timebased_dynamic (0  , ( 10UL )         , _LCDML_start  , LCDML_BACKEND_control);

or 

LCDML_BACK_new_timebased_dynamic (0  , ( 5UL )         , _LCDML_start  , LCDML_BACKEND_control);

I will rewrite this part on weekend. At the moment i cannot find my encoder :-(

StafWars, 2016-05-26:
Thanks for the fast respons. 
But when I try this, it doesn't compile. Or do I  have to insert it on a library file?

Kind regards

StafWars

[image]

closed 2017-01-08 by Jomelo