Jomelo/LCDMenuLib #17: How to put names in menu

jcmojj, 2016-10-27:
You did a great job with that library. I spent this week understanding it because the documentation is in German.  Now that I figured it out, I just need to do one thing that I didn't understand the best way to do it. I need to do a menu that is not exactly dynamic as in your example. In your dynamic menu example you basically make a menu inside the last screen of you normal static menu. What I need is to use specific words in the main menu so I can acess info regarding that menu. Just for example imagine I have 6 people using my arduino to mesure their weight and save it. Each person has a name that could be added to that menu. So the menu would be the name of each person. I will save it in the EEPROM. Everyday the people will measure their weight and save it at the eeprom ( for 10 days). So at the main menu it will be the name of the people. The sub-menu of each person  will be to change their name, to show their average weight in the last 10 days, and so, and I can do all that. I just don't know how to show some eeprom string (that could be even some normal string to make your example easier) at the menu. It is not a dynamic menu, otherwise I would need to make the whole menu dynamic and it is no point to do so if I can use the power of your menu ( by far the best menu library I found)!

manosv, 2016-10-27:
I second that.I have the same problem, i want to do something like that but i dont know how.Maybe the developer knows a way.Your library is fantastic by the way.

Jomelo, 2016-10-27:
Hi, 
i create a new version 2.1.5. 
In this version i add a new public variable witch contains the content_id from init. 
Now you can make following in the "LCDML_DISP" tab.

//..
// set cursor
lcd.setCursor(1, n);
// set content
lcd.print(LCDML.content[n]);
// new
if(LCDML.content_id[n] == 2) {
  lcd.print(" ");
  lcd.print(global_variable_content);
}

I think this is the shortest way to help you :-) 
Are there more improvements for this lib ?

closed 2017-01-08 by Jomelo