Jomelo/LCDMenuLib #20: Time in menu

dreep, 2016-12-02:
Hello everybody,

Very nice job with this library.
I would like to insert date and time as a menu title... can we do that and how ?

Thx for help !

dreep, 2016-12-23:
Nobody to help me ?
I can't read German for complete instruction and i don't find how i can display date and time (with RTC) in the menu... So, i would like to replace one item ont the first layer to show real time...
I try somethings but without results...

Plz help...
Thx a lot.

Jomelo, 2016-12-23:
I Look on it after Christmas. At the moment i have no time.

dreep, 2016-12-24:
Thank you very very much !!
To help you to understand what i am looking for :
`char date_time = "00/00/00 00:00:00";
LCDML_DISP_init(_LCDML_DISP_cnt);
LCDML_DISP_add      (0  , _LCDML_G1  , LCDML_root      , 1  , date_time       , LCDML_FUNC_p2);/*_AFF_DATE*/
LCDML_DISP_add      (1  , _LCDML_G1  , LCDML_root      , 2  , "Arret d urgence"    , LCDML_FUNC_p2);/*ARRET_URG*/
LCDML_DISP_add      (2  , _LCDML_G1  , LCDML_root_1      , 1  , "Annuler"            , LCDML_FUNC_p2);/*_RETOUR*/
LCDML_DISP_add      (3  , _LCDML_G1  , LCDML_root_1      , 2  , "TOUT ARRETER"       , LCDML_FUNC_p2);/*_STOP*/
LCDML_DISP_add      (4  , _LCDML_G1  , LCDML_root      , 3  , "Les moteurs"        , LCDML_FUNC_p2);/*_RESUM_MOT*/
LCDML_DISP_add      (5  , _LCDML_G1  , LCDML_root_3      , 1  , "Moteur 1"           , LCDML_FUNC_p2);/*_RES_MOT1*/`

On the first line, i would like to use a variable (date_time) which i update with my RTC shield...
I hope that's clear for you...

Have a good christmas and you do a very nice work with this library which help a lot !!! Very nice work !

Jomelo, 2017-01-08:
I have tested it and i create a new version next week. With the last version you can add dynamic content, but you can not update this content every secound. I create a new makro wich do this job.

Jomelo, 2017-01-08:
Update is online 
Please check the  example "LCDML_dynamic_menuitem_text"

dreep, 2017-01-08:
Thx a lot for your job and Time !
On the same time, i convert your lib with i8g2 and i leave thé master row of my screen free... And add drawstr(0,40,time); u8g2.sendbuffer; on my l'opportunité and it works fine !
But I will table à look on your updates beacause i think you dis something much betteraves than me !

closed 2017-01-12 by Jomelo