Jomelo/LCDMenuLib #21: u8g2 examples

AlexFundorin, 2017-01-08:
Please, add compatibilty (examples of how to use LCDMenu) with new version of u8g library.
The new library can be found here - https://github.com/olikraus/U8g2_Arduino

Jomelo, 2017-01-08:
I check this tomorrow.
When i have read this 3 hour early i have added it today :-)

AlexFundorin, 2017-01-08:
Thanks alot, cause it's time for my project to obtain a neat menu system, which defenitely would be your library. :)

Jomelo, 2017-01-10:
https://github.com/Jomelo/LCDMenuLib/tree/master/examples/LCDML_DISPLAYTYPE/LCDML_glcd_u8g2lib

Please check the new example. I cannot check the complete functional because the 8ug2lib needs many ram. On my small Arduino Duemilanove 105 %

I reduce the test program to only 5 menuitems, then it works.

AlexFundorin, 2017-01-10:
> examples/LCDML_DISPLAYTYPE/LCDML_glcd_u8g2lib

I've only got this compiling error, but the example uploaded fine to my
Teensy++ 2.0 board.

I think that the example should work, though I don't have any regular
buttons on my device to test it. Only capacitive pads are available, which
require additional code to become usable. I'll try to merge the provided
example to my current project tomorrow and will report after it's done.
Thank you for the update. There are different modes in u8g2. Full buffer,
page buffer and 8x8. The first one takes most of the ram, while the last
one takes least. Or is it not affecting RAM usage at all?
[image: Встроенное изображение 1]

Jomelo, 2017-01-10:
Ok, i will test the other modes on weekend.

AlexFundorin, 2017-01-11:
Is it possible to make icons (3x2) menu with your library, where each icon
is an ASCII symbol or user drawn bitmap (u8g2 lib allows to draw bitmaps,
using arrays)?
The other thing that I'm interested in is to be able to display current
selected icon's name at the top of the display. And the last thing that is
interesting to me is to be able to have regular corresponding menu when any
one of the six title icons was selected.
Something like this but also with current selected item description at the
top or bottom of the screen -
http://www.androidheadlines.com/wp-content/uploads/2014/02/AH-Smartwatch2-Review-5-e1395239728285.jpg

If this isn't possible, than, maybe, a set of icons at the top of the
screen (top menu level) and submenu items displayed at the rest of the
screen. Like here:
https://qph.ec.quoracdn.net/main-qimg-a2cbf7b26236b5e0005507eb9e8dc95a?convert_to_webp=true

Jomelo, 2017-01-12:
Hi, 

**icons (3x2):** it is possible, you have to change the LCDML_DISP Tab content for your application, but i have no time to write it for you

**selected item on top:**  the menu is cursor based and i have to change many things in this lib to make this. It is an idea for version 3.0.0 when i have to change some interfaces. 

tob bar of icons: it is possible, you have to rewirte the LCDML_DISP Tab completly. You can get the parent items to diplay the topbar. But it needs many time. -> 3.0.0

I update today or tomorrow the u8glib and u8g2lib examples. There is a bug with the scrollbar handling. With this update i add the "box mode" withit it is possible to set a box around the menu like width=50, height=70 on position x0=10, y0=10.

closed 2017-03-29 by Jomelo