Jomelo/LCDMenuLib #29: Execute display and backend functions parallel

dragoblaztr, 2017-04-18:
Is it possible to run backend and display functions (in my case dinamic menu item text) ??, I need to run a backend function that run a code but this stops the display functions, is there any way to run them in parallel.

Regards

Jomelo, 2017-04-18:
That the Backend Menü Funktion stops, is normal.  You can use triggers to Run this functions every x millisecounds. Or you can create a new Backend function. Look into the backend example.

Sorry, for this bad englisch, this keyboard have an autocorrection for another language :( ..

dragoblaztr, 2017-04-18:
I tried with triggers getting same results

How supposed to be the order or execution... because no matter what order is, always is executed the backend function before the display func

`// enter file name
      if(LCDML_BUTTON_checkEnter()) {
        LCDML_BUTTON_resetAll();   
        LCDML_DISP_funcend();
        LCDML_DISP_jumpToFunc(LCDML_FUNC_infoscreen);  //this is the function that stops or executed
        Serial.println(filelist[cursor_position_cur]);                    //after the backend ends
        LCDML_BACK_start(LCDML_BACKEND_static);
        }`

Regards

Jomelo, 2017-04-19:
I hope i understand you correctly.
Look in "LCDML_serialmonitor" example on tab "LCD_FUNC_DISP" on line 78  (Function: void LCDML_DISP_setup(LCDML_FUNC_timer_info))

There is in the setup function a trigger:
 LCDML_DISP_triggerMenu(100);  // starts a trigger event for the loop function every 100 millisecounds

In the Loop function the initscreen timer have to be reset with:
 // reset the initscreen timer
  LCDML_DISP_resetIsTimer();

Now, the loop function is called every 100 ms until you get your exit signal.

`void LCDML_DISP_setup(LCDML_FUNC_timer_info)
// *********************************************************************
{
  // setup function
  Serial.println(F("==========================================="));
  Serial.println(F("================  FUNC ===================="));
  Serial.println(F("==========================================="));
  Serial.println(F("wait 10 secounds or press back button"));     
  g_func_timer_info = 10;       // reset and set timer    
  LCDML_DISP_triggerMenu(100);  // starts a trigger event for the loop function every 100 millisecounds
}

void LCDML_DISP_loop(LCDML_FUNC_timer_info)
{ 
  // loop function, can be run in a loop when LCDML_DISP_triggerMenu(xx) is set
  // the quit button works in every DISP function without any checks; it starts the loop_end function 
  
  // this function is called every 100 millisecounds
  
  // this timer checks every 1000 millisecounds if it is called
  if((millis() - g_timer_1) >= 1000) {
    g_timer_1 = millis();   
    g_func_timer_info--;                // increment the value every secound
    Serial.println(g_func_timer_info);  // print the time counter value     
  }
  
  // reset the initscreen timer
  LCDML_DISP_resetIsTimer();
  
  // this function can only be ended when quit button is pressed or the time is over
  // check if the function ends normaly
  if (g_func_timer_info <= 0)
  {
    // end function for callback
    LCDML_DISP_funcend();  
  }   
}`

dragoblaztr, 2017-04-19:
I tried that but, I got the same results...

That I try to do is:
first place load a dynamic submenu loading the files stored in a sd card (done).
second select file and processing to send it trought a serial port (serial1) for use in another arduino (done)
third jump to an info screen containing dinamic text parallel at processing on 2nd step (in process)

but when I send the file for processing the display functions stops and no longer avaliable until backend finishes processing, even I sent display functions prior backend functions as in the example

`// enter file name if(LCDML_BUTTON_checkEnter()) { //********************************* LCDML_BUTTON_resetAll(); //supposed to be the 1st instruction******************       
LCDML_DISP_funcend();////supposed to be the 2nd instruction****************  LCDML_DISP_jumpToFunc(LCDML_FUNC_infoscreen); ////supposed to be the 3rd instruction Serial.println(filelist[cursor_position_cur]); //supposed to be the 4th instruction LCDML_BACK_start(LCDML_BACKEND_static);//supposed to be the last instruction }`

No matter how i declare it, always execute first the backend functions and after it finishes the rest of code are done, even I tried to add a delay before the backend function with the same execution order (first backend then the rest including the delay).

Or am I wrong??

Regards

PD. If you need my complete source files I will upload for a near looking...

dragoblaztr, 2017-05-11:
This is my function for read and streaming the g code from SD card to a second arduino throught the serial1

```
 // *********************************************************************
File myFile; 
void LCDML_BACK_setup(LCDML_BACKEND_static)
// *********************************************************************
{
  // setup
  // is called only if it is started or restartet (reset+start)
  //LCDML_DISP_update_menu();
  Serial1.begin(115200);
  myFile = SD.open(gFolder + filelist[cursor_position_cur], FILE_READ);    //create a new file
}
boolean LCDML_BACK_loop(LCDML_BACKEND_static)
{
    LCDML_DISP_update_menu();
    //LCDML_DISP_resetIsTimer(); // reset the initscreen timer
    //LCDML.goBack();            // go back
    //LCDML_DISP_funcend();      // LCDML_DISP_funcend calls the loop_end function
    //LCDML_DISP_update_menu_direct(0);
    sendGcode();
  }

void emptySerialBuf(int serialNum){

  //Empty Serial buffer
  

  if(serialNum==0){
    while(Serial.available())                      
      Serial.read();
  }
  else if(serialNum==1){
    while(Serial1.available())                      
      Serial1.read();
    }
}

void waitSerial(int serialNum){

  // Wait for data on Serial
  //Argument serialNum for Serial number

  boolean serialAv = false;

  if(serialNum==0){
    while(!serialAv){ 
      if(Serial.available())
        serialAv=true;
    }
  }
  else if(serialNum==1){
    while(!serialAv){ 
      if(Serial1.available())
      serialAv=true;
    }
  }
}

String getSerial(int serialNum){

  //Return String  from serial line reading
  //Argument serialNum for Serial number

  String inLine = "";
  waitSerial(serialNum);

  if(serialNum==0){
    while(Serial.available()){              
      inLine += (char)Serial.read();
      //wait(2);
    }
    return inLine;
  }
  else if(serialNum==1){
    while(Serial1.available()){               
        inLine += (char)Serial1.read();
        //wait(2);
    }
    return inLine;
  }
}

void sendGcode(){

  //READING GCODE FILE AND SEND ON SERIAL PORT TO GRBL

  //START GCODE SENDING PROTOCOL ON SERIAL 1

  String line = "";

    Serial1.print("\r\n\r\n");      //Wake up grbl
    //wait(2);
    emptySerialBuf(1);
    if(myFile){                                                                      
      while(myFile.available()){    //until the file's end
        line = readLine(myFile);  //read line in gcode file 
          Serial.print(line);   //send to serials
          Serial1.print(line);
          Serial.print(getSerial(1)); //print grbl return on serial
      }
    }
    else
      fileError();

  myFile.close();
  LCDML_BACK_stop(LCDML_BACKEND_static);
  //LCDML_BACK_stopStable(LCDML_BACKEND_static);
  LCDML_BACK_reset(LCDML_BACKEND_static);
  //LCDML_BACK_dynamic_restartTimer(LCDML_BACKEND_static);
  //LCDML_BACK_all_reset();
  Serial.println("Finish!!\n");
  //wait(2000);
}

void fileError(){

  // For file open or read error

  Serial.println("\n");
  Serial.println("File Error !");
}

String readLine(File f){
  
  //return line from file reading
  
  char inChar;
  String line = "";

  do{
    inChar =(char)f.read();
      line += inChar;
    }while(inChar != '\n');

  return line;
}

void LCDML_BACK_stable(LCDML_BACKEND_static)
{
  // stable stop
  // is called when a backend function is stopped with stopStable
  Serial.println(F("Stop Stable: LCDML_BACKEND_static"));
  Serial.println(F("================================="));  
}

```

I created  a function that it's similar to a delay but without delays using elapsedMillis library

```
  void wait(unsigned long alfa){
  elapsedMillis waop;
  while(waop < alfa){
    }
  }
```

And the results the same, until the streaming ends the remaining functions are executed, even the higher priority than this.

Any sugestions...

Jomelo, 2017-05-11:
Hmm,
you can replace your wait function when you are using serial event  https://www.arduino.cc/en/Tutorial/SerialEvent

I think the program runs for the complete stream in a while loop and cannot run the other functions. 
Can you show me your backend initialisation ?

dragoblaztr, 2017-05-11:
it's this:

```
// ********************************************************************* 
// LCDML BACKEND (core of the menu, do not change here anything yet)
// ********************************************************************* 
  // define backend function  
  #define _LCDML_BACK_cnt    2  // last backend function id
  
  LCDML_BACK_init(_LCDML_BACK_cnt);
  LCDML_BACK_new_timebased_dynamic (0  , ( 1000UL )  , _LCDML_start  , LCDML_BACKEND_sec_counter);
  LCDML_BACK_new_timebased_dynamic (1  , ( 20UL )    , _LCDML_start  , LCDML_BACKEND_control);
  LCDML_BACK_new_timebased_dynamic (2  , ( 1000UL )  , _LCDML_stop   , LCDML_BACKEND_menu);
  LCDML_BACK_new_timebased_dynamic (3  , ( 5000UL )  , _LCDML_stop   , LCDML_BACKEND_static);
  LCDML_BACK_create();
```

Regards

dragoblaztr, 2017-05-12:
Could you do a sugestion to read and stream the gcode ? without interfering of the execution of the rest of functions?

Regards

Jomelo, 2017-10-14:
i think this is done with v3.0.0
The next development version is v3.0.0 beta 1

Jomelo, 2017-10-25:
https://github.com/Jomelo/LCDMenuLib2

The backend is removed in this version. You can use an alternative task system. 
Place the LCDML.loop_control() function in a fast task and the LCDML.loop_menu() function in a slow task. The LCDML.loop() is not needed.
When you place your gcode stream in a high priority task, it have to work.

closed 2017-10-25 by Jomelo