Jomelo/LCDMenuLib #31: Dublicate code

nachaliuga, 2017-05-12:
All values change, everything is done similar to the example 
I will leave video with a problem  I somebody can help :)
https://www.youtube.com/watch?v=CE-gpdlapgk

dragoblaztr, 2017-05-12:
@nachaliuga 
Can you upload your code for give it a near looking?

nachaliuga, 2017-05-13:
http://forum.arduino.cc/index.php?topic=476638.0
there are alot stupid stuff over there  if you need full code let me know :)

Jomelo, 2017-05-13:
Thanks, i will have a look on it, when i am at home on sunday. At the moment i have no access to any PC. (Only Handy)

dragoblaztr, 2017-05-13:
@nachaliuga 
Can you post your main ino code? on forum you only show your disp functions

Regards

Jomelo, 2017-05-14:
@nachaliuga 
I need your main code. I think you have a bug in the initialisation.

nachaliuga, 2017-05-15:
https://ufile.io/8dbpy 
here it is, it was a mistake that led to value change for both but now another problem is that cursor doesnt want to move on the other window I realy appreciate your help

Jomelo, 2017-05-16:
What happens, when you change your control method to serial ? 
What mean you with "other window", which window ?

nachaliuga, 2017-05-17:
I'm idiot forget it case is closed  :D

closed 2017-05-17 by Jomelo