Jomelo/LCDMenuLib #35: Change parameter at menu item

dragoblaztr, 2017-08-29:
How do I change parameter directly on any menu item

like
Main
   feature1 (here to enable or disable directly without jumping at submenu, just left or right)
   feature2 (here change value without jump to another submenu just left or right)

And so on
[image]
[image]

dragoblaztr, 2017-08-29:
https://www.youtube.com/watch?v=9UAlEIZfd-Y

Jomelo, 2017-08-30:
Hello,
on a direct way it is not possible. But you can open a menufunction and change a global variable. 

Then you can open the "LCDML_dynamic_menuitem_text" example and open the "LCDML_DISP" tab. On line 32 you can find this code.
`// set content
		// with content id you can add special content to your static menu or replace the content
		// the content_id contains the id wich is set on main tab for a menuitem
        switch(LCDML.content_id[n])
		{		 
			case 0:
				lcd.print("Sec counter: "); // or datetime or other things
				lcd.print(g_sec_counter);
				break;
		
			default: // static content
				lcd.print(LCDML.content[n]);
				break;				
} `

You can add here another case option with your menu item id to change the content and display your custom chars.

Jomelo, 2017-10-14:
I add this function to version 3.0.0. Then it is possible to change content with the button when the cursor stands on an element.

dragoblaztr, 2017-10-15:
So much tanks

I have a question, version 3.0.0 is released and stable or still under development?

Jomelo, 2017-10-15:
It is a development versio. I try to release the first beta in some hours. 
There are some small bugs at the moment:
- jump to hidden submenues does not work correctly 

ToDo: 
- rename LCDML_add to LCDML_addAdvanced when all callback functions are used
- add a callbackfunction for the screensaver
- add error codes for some debug errors
- some changes on the backend system (backend and lcdml can use as standalone)

When i add a tag with "beta" i won`t change the functionnames and fix only bugs or add example.

Jomelo, 2017-10-15:
You can test the beta

dragoblaztr, 2017-10-16:
Do you have an scheduled date for a new stable version?

Jomelo, 2017-10-16:
Not at the moment. The library is only a hobby :-)

dragoblaztr, 2017-10-20:
Another question, it will work on arduino due or others arm32 mcus or need to modify sources? if so hard to do?

Jomelo, 2017-10-20:
I haven't test it. 
I think you need a c++ compiler for the other controller and define some data types, set the "use ram" parameter in "LCDMenuLib.h".

Jomelo, 2017-10-20:
I haven't test it. 
I think you need a c++ compiler for This  other controller and define some data types, set the "use ram" parameter in "LCDMenuLib.h".