Jomelo/LCDMenuLib #37: Millis() rollover handling

dragoblaztr, 2017-09-15:
This library has support to handling of millis() rollover or needs to be modified to do that?

If not has to be updated by modifing the cpp file and sketches also?

Regards

Jomelo, 2017-09-15:
It is supported.
You can find the Code in LCDMenuLib_macros.h

'
// macro: thread timer with return
	// attention: when the wait_time is bigger then millis on startup the result ist true 
	// to fix this, the timer_var have to be initialise with the waittime 
	#define LCDML_BACK_TIMER(timer_var, wait_time)\
		if(!((millis() - timer_var) >= wait_time)) {  return; }\
		timer_var = millis();
'

dragoblaztr, 2017-09-15:
So the other functions that use millis() it does to be modified with something like that or not?

because I assumed that it doesn't support it and i made this fix (by example in one one my functions)

`if((unsigned)(millis() - g_lcdml_initscreen) >= _LCDML_DISP_cfg_initscreen_time) {`

Jomelo, 2017-09-15:
I will check the examples as soon as possible. 
The "backend" function use the "timer" macro.

closed 2017-10-14 by Jomelo