Jomelo/LCDMenuLib #40: Jump between disp functions

dragoblaztr, 2017-10-02:
it's possible to jump between display functions, becase I tried and don't get expected results.

Per example I'm in a custom display function and with a user input control y want to jump to another display function and it doesn't do that (except by inis screen to another disp func but viceversa neither).

I used similar commands like:

```

LCDML_DISP_funcend();
LCDML_DISP_jumpToFunc(LCDML_FUNC_initscreen);
```

or viceversa and it work neither

Help please

Jomelo, 2017-10-03:
I think this is a bug. 
In older versions it works, but in the last version and in the "development" version it did not work.

dragoblaztr, 2017-10-03:
Do you have an idea how to fix it, without using older versions, or display functions have big changes in that functions?

Regards

Jomelo, 2017-10-03:
I try it tomorrow. 
You use the latest stable version 2.3.2?

dragoblaztr, 2017-10-03:
Yes it is, what build has that working properly for make a test?

Jomelo, 2017-10-03:
I don`t know. 
I have no automatic test system for all possible configuration.

dragoblaztr, 2017-10-03:
I starting try at 2.1 (cross fingers) and let see. Besides I stay pending at your answere .

Regards

dragoblaztr, 2017-10-03:
Not working 2.2, 2.1 or 2.0 neither.

Jomelo, 2017-10-04:
Ok thx for testing. 
I can reproduce the problem and i hope i can fix it on friday.

dragoblaztr, 2017-10-05:
I found this difference between  versions

```
v1
/* ************************************************ */
/* LCDMenuLib_jumpToFunc							*/
/* ************************************************ */
#define LCDMenuLib_jumpToFunc(name)\
	for(uint8_t i=0; i<=255;i++) {\
		if(name == g_LCDMenuLib_functions[i]) {\
			LCDML.jumpToElement(i);\
			break;\
		}\
	}\
	return;

v3	
#define LCDML_DISP_jumpToFunc(name)\
	for(uint8_t i=0; i<_LCDML_NO_FUNC;i++) {\
		if ( name##_loop_setup == g_LCDML_DISP_functions_loop_setup[i]) { \
			if(LCDML.getFunction() != _LCDML_NO_FUNC) {\
				g_lcdml_jump_func = i;\
			} else {\
				LCDML.jumpToElement(i);\
				LCDML_DISP_update_menu(); \
			}\
			break;\
		}\
	}\
	LCDML_BUTTON_resetAll()
	
v2	
#define LCDML_DISP_jumpToFunc(name)\
	for(uint8_t i=0; i<=254;i++) {\
		if (name##_loop_setup == g_LCDML_DISP_functions_loop_setup[i]) { \
			if(LCDML.getFunction() != _LCDML_NO_FUNC) {\
				bitSet(LCDML.control, _LCDML_control_funcend);\
				g_lcdml_jump_func = i;\
			} else {\
				LCDML.jumpToElement(i);\
				LCDML_DISP_update_menu(); \
			}\
			break;\
		}\
	}\
	LCDML_BUTTON_resetAll()
	
```
Maybe this can help you

Thanks in advance

dragoblaztr, 2017-10-06:
Any progress??

Jomelo, 2017-10-07:
Can you check the indexing of your function in the main tab ?

'  
// this value must be the same as the last menu element
  #define _LCDML_DISP_cnt    11
'

It don't work when the jump function is called before the initialisation is complete.

I test it with the "serial_monitor" example and it works.
'
void LCDML_DISP_loop(LCDML_FUNC_information) 
{
  // loop function, can be run in a loop when LCDML_DISP_triggerMenu(xx) is set
  // the quit button works in every DISP function without any checks; it starts the loop_end function   
  if(LCDML_BUTTON_checkAny()) { // check if any button is presed (enter, up, down, left, right)
    // LCDML_DISP_funcend calls the loop_end function
    LCDML_DISP_funcend();
    
    LCDML_DISP_jumpToFunc(LCDML_FUNC_p2);
  } 
}
'
 

It can be happen, that the menu is shortly displayed before the next function is called. I will fix this with the next version.

dragoblaztr, 2017-10-07:
I tried it by mi code adding the serial print and effectively it jump how it supposed to be and kickly to next function.

So the only thing can I do it's waiting for the new version? There's no kick fix for this?

Jomelo, 2017-10-08:
Can you send me your complete code with the jump function which don`t work ?
( nilsfeld (at) gmx (dot) de )

dragoblaztr, 2017-10-11:
I sent to you the code, do you hace any chance of give it a look?

dragoblaztr, 2017-10-20:
Hi again, could be possible to use the LCDML_DISP_jumpToFunc macro with any parameter?

Cheers

Jomelo, 2017-10-20:
You can use this macro only with the function name. For example:

LCDML_DISP_jumpToFunc(your_menu_function_name); // jump to your menu function name

dragoblaztr, 2017-10-20:
what if that function has parameters like these
`
LCDML_DISP_addParam (2  , _LCDML_G3  , LCDML_root_1      , 2  , "Dis Program"        , LCDML_FUNC_program_change, 1);`

how to send parameter?

Jomelo, 2017-10-21:
No Problem, 
i try to write a better documentation for the next version. Your questions show me, that the documentation is very bad :-(

You can get the parameter from the activ function with:
LCDML_DISP_getParameter();

The callback function are saved in an array of function pointern. You can call it with:

0,1,2 = your function ids

g_LCDML_DISP_functions_loop_setup[0]();
g_LCDML_DISP_functions_loop[0]();
g_LCDML_DISP_functions_loop_end[0]();

In the next version all the arrays are placed in the item object.

Jomelo, 2017-10-25:
Next verson is stable
https://github.com/Jomelo/LCDMenuLib2

A simple documentation can be found in the wiki

closed 2017-10-25 by Jomelo

dragoblaztr, 2017-10-25:
Superb thank you

fietstasss, 2018-01-09:
I'd like to call a void/function when a menu item is clicked/selected, but with a parameter. 
The void will set a specific effect/pattern to a led strip, the effects are summed in an array and can be called by a number. 
In example: The menu exists off: 

> Effects
> ---- MovingEffect
> ---- FadingEffect
> ---- JumpingEffect
>  Brightness
>  Speed

  Then, when clicking on for example 'Fadingeffect', a function is called which runs something like: 
SetPattern(2); 

Now, is it possible to implement the number of the effect (2 in this case), directly in the menu structure functioncall ? 

Hope my question is clear. 

Thanks in advance.

Jomelo, 2018-01-09:
Have you looked into the "AVR -> LCDML_DISP -> LCDML_with_parameter" example ?

When you would change the parameter per hand you have use this:
g_lcdml_param[_your_menu_elment_id_] = 120; // max is 255 (one byte)

fietstasss, 2018-01-09:
Thanks, that did the trick!
Can you also help me on how to change a variable when in a submenu? 
For instance when i click Brightness, i've now made a menu which shows the current Brightness value, but i'd also like to be able to adjust it by rotating the rotary encoder when this submenu is active. 

The information is currently displayed by: 

`    void LCDML_DISP_loop(BrightnessSet)
    { 
      // loop
      // is called when it is triggert
      // - with LCDML_DISP_triggerMenu( millisecounds ) 
      // - with every button status change

u8g.firstPage(); 
 do {
u8g.setFont(_LCDML_DISP_font);
u8g.drawStr(15, 20, "Brightness: ");
  
u8g.setFont(_LCDML_DISP_fontLarge1);
//u8g.drawStr(8, 30, "Eeee");

char buf[9];
sprintf (buf, "%d", BRIGHTNESS);
u8g.drawStr(25, 55, buf);

  } while( u8g.nextPage() );
  delay(200); 

      // check if any button is presed (enter, up, down, left, right)
      if(LCDML_BUTTON_checkAny()) {         
        LCDML_DISP_funcend();
      } 
    }`

Thanks for your great help!

Jomelo, 2018-01-10:
I hope the delay(200) is only a placeholder, when not please remove it and do not use it agian in a menu function. (look on the "blink without delay" example)

-> change the value in a submenu => dynamic content example
-> change the value in a submenu function => LCDML_change_value example

fietstasss, 2018-01-10:
Thanks for your reply Jomelo!

I’ve managed to change the value within the menu, only thing is, after every turn of the encoder/every change, the menu closes and has to be reopened again, otherwisely said: The ‘quit’ event is called after every adjustment in brightness value.
Can you tell me why/where this comes from/how to prevent it?

The code beneath is used to adjust the brightness.

Thanks in advance!

   void BrightnessChange(int AdjustVal){
//      TempBrightness

      BRIGHTNESS = BRIGHTNESS + AdjustVal;
      FastLED.setBrightness(BRIGHTNESS);
      BrightnessDisplay();
    }

    void BrightnessDisplay(){
            ////////////Void to print brightness value to the screen //////////////
            // loop
           // is called when it is triggert
           // - with LCDML_DISP_triggerMenu( millisecounds )
           // - with every button status change

            u8g.firstPage();
            do {
            u8g.setFont(_LCDML_DISP_font);
            u8g.drawStr(15, 20, "Brightness: ");

            u8g.setFont(_LCDML_DISP_fontLarge1);
            //u8g.drawStr(8, 30, "Eeee");

            char buf[9];
            sprintf (buf, "%d", BRIGHTNESS);
            u8g.drawStr(25, 55, buf);

             } while( u8g.nextPage() );
             //delay(200);

           // check if any button is presed (enter, up, down, left, right)
           if(LCDML_BUTTON_checkAny()) {
             LCDML_DISP_funcend();
           }

    }

    ///////////////////////////////BrightnessSet/////////////////////
    void LCDML_DISP_setup(BrightnessSet)
    {
      // setup
      // is called only if it is started
            digitalWrite(3, HIGH);

            BrightnessDisplay();

      // starts a trigger event for the loop function every 100 millisecounds
      LCDML_DISP_triggerMenu(100);

    }

    void LCDML_DISP_loop(BrightnessSet)
    {
      // loop
      // is called when it is triggert
      // - with LCDML_DISP_triggerMenu( millisecounds )
      // - with every button status change

      // check if any button is presed (enter, up, down, left, right)
      if(LCDML_BUTTON_checkAny()) {
                if (LCDML_BUTTON_checkUp())    { BrightnessChange(+40); }//LCDML_BUTTON_resetUp }
                if (LCDML_BUTTON_checkDown())  { BrightnessChange(-20); }//LCDML_BUTTON_resetDown}
                if (LCDML_BUTTON_checkEnter())  { BrightnessChange(-20); }//LCDML_BUTTON_resetDown}

//if (LCDML_BUTTON_up())    { HELP_FUNC_set_dat(+1); BrightnessChange(1); }
//if (LCDML_BUTTON_down())  { HELP_FUNC_set_dat(-1);  BrightnessChange(-1); }

//                if (LCDML_BUTTON_quit()) {LCDML_DISP_funcend(); }
//                if (LCDML_BUTTON_enter()) {LCDML_DISP_funcend(); }

//        LCDML_DISP_funcend();
      }
    }

    void LCDML_DISP_loop_end(BrightnessSet)
    {
      // loop end
      // this functions is ever called when a DISP function is quit
      // you can here reset some global vars or do nothing
    }

dragoblaztr, 2018-01-10:
@fietstasss 

Your problem is here

```
           // check if any button is presed (enter, up, down, left, right)
           if(LCDML_BUTTON_checkAny()) {
             LCDML_DISP_funcend();
           }
```
Get your own conclusions.

Regards

fietstasss, 2018-01-10:
Thanks! Fixed that one!

Last question i guess, the current rotary encoder handling in the library examples isn't working with my mechanical encoder very well. I've found this example code using interrupts, which works flawlessly with my encoder: 
http://www.instructables.com/id/Improved-Arduino-Rotary-Encoder-Reading/

I tried implementing it in my existing code but unfortunately didn't have any succes yet, does anyone have an idea on how to implement this into the existing LCDMenuLib ?

Thanks in advance!

dragoblaztr, 2018-01-10:
Check this out (at end)

[https://github.com/Jomelo/LCDMenuLib/issues/14#ref-issue-258467553](url)

Regards