Jomelo/LCDMenuLib #41: _LCDML_priority how does it work?

paulosilver, 2017-10-24:
hello how does _LCDML_priority work? 
How can o give priority to a backend function?
thank you

Jomelo, 2017-10-25:
It worked like a list with an option to reset the list counter. When you use LCDML_without_priority the option to reset the list does not exists. 

The priority has no real-time effekt. 

http://jomelo.de/r/arduino/lib_simpleThread/ablauf.png

paulosilver, 2017-10-25:
ok, so if i want to add priority to a function what do i have to do?

Jomelo, 2017-10-25:
You do it at the initialisation with the id number 0,1,2,3 ...

paulosilver, 2017-10-26:
[image]
here?

Jomelo, 2017-10-26:
Yes, but the LCDML_BACKEND_stepper1_run and LCDML_BACKEND_stepper1_accelMov have a special return value. When the time is set to 0 and the return value has the wrong value (false), then it do not work correctly. 

In the new version of the Lib i removed the backend system because it is not a direct part of the LCDMenuLib
The new LCDMenuLib2 version can be downloaded here: https://github.com/Jomelo/LCDMenuLib2 
(all Makros and all function names have been changed)
Currently i work at the new documentation: https://github.com/Jomelo/LCDMenuLib2/wiki/Code-Reference

You can use this lib:  https://github.com/arkhipenko/TaskScheduler

paulosilver, 2017-10-26:
ok i am going to change to that version
thank you

closed 2017-10-26 by paulosilver