Jomelo/LCDMenuLib #44: Rotary encoder setup

fietstasss, 2017-12-28:
Hey, 

I'm currently using this library's example to create a menu, but unfortunately can't get my rotary encoder to work with it. 

My two questions about it are: 
1. Which way should the Rot.Encoder be wired up? Currently i'm using pin 10,11,12 as I think I've seen this in the code, is this correct? (10 as pin A, 11 as pin B, 12 for the switch output, switch is connected to 5V, common pin connected to ground.)

2. What do i have to edit/adjust in order to be able to use the rotary encoder with this library? 
Looking forward to your reply. 

A. 

Also crossposted on:
https://forum.arduino.cc/index.php?topic=519339.0

Jomelo, 2017-12-28:
change _LCDML_CONTROL_cfg  from 0 to 3

closed 2017-12-28 by Jomelo

fietstasss, 2017-12-30:
Dear,

Do you also know any way to use a longpress on the encoder switch as a ‘go-back’ action?
Can’t seem to find out how to implement this.

Thanks in advance.

A.

dragoblaztr, 2018-01-19:
This setup  https://github.com/Jomelo/LCDMenuLib/issues/14 is able to do that, at this time for go back button is used double click but its easy to change this line

`button.attachDoubleClick(quit);`

to this

`button.attachLongPressStop(quit);`

and that's it.

Regards