Jomelo/LCDMenuLib #47: Difference between LCDML and LCDML2

alimnette, 2018-01-26:
Hello,
first of all thank you for your work.

What is the difference between LCDML and LCDML2?

Thank you

Jomelo, 2018-01-26:
Hello,

the menu structure is the same.

LCDML 2:
- remove backend system from LCDML 1
- (you can use every other task system)
- remove group system (8 groups max) from LCDML 1 
- add a condetion system (with no limit) 
- remove over 50 makros from LCDML 1 because i cannot add new functions with it 
- rewrite the complete class to use more methods insteat makros
- add an easy way to change menuitems / parameters direct in the menu

LCDML V1:
- i'm not adding new function in future ( only bug fixes )
- the group system with hidden buttons works not correctly
- the function "jumpToElement" works not correctly

alimnette, 2018-01-28:
Hello,

OK thanks for your answer.
It is better to use the V2

closed 2018-01-28 by Jomelo