Jomelo/LCDMenuLib #53: Question by definition #define LCDML_DISP_addMenu

DmitryGoldobin, 2019-04-29:
In definition 

#   define LCDML_DISP_addMenu(name, disp, item_parent, item_child, content)\
...
      void function ## _loop_setup(); \
      void function ## _loop(); \
      void function ## _loop_end(); \

at the same time, it is not defined "function".

Please explain how it works or it is a bug and instead of a "function" it should be "LCDML_FUNC" ?
(Sorry for bad English)

Jomelo, 2019-04-29:
Hello, 

my english is not better ;-)

It is not a bug. The code shows a part of a macro.

```
#		define LCDML_DISP_addMenu(name, disp, item_parent, item_child, content)\
			const char g_LCDML_DISP_lang_ ## name ##_var[] = {content};\
			void function ## _loop_setup(); \
			void function ## _loop(); \
			void function ## _loop_end(); \
			LCDMenu item_parent ## _ ## item_child(name, disp); \
			void LCDML_DISP_##name##_function() { g_LCDML_DISP_functions_loop_setup[name] = LCDML_FUNC_loop_setup;  g_LCDML_DISP_functions_loop[name] = LCDML_FUNC_loop; g_LCDML_DISP_functions_loop_end[name] = LCDML_FUNC_loop_end; item_parent.addChild(item_parent ## _ ## item_child); }
```
Look on this links:  
- https://en.wikipedia.org/wiki/C_preprocessor
- https://www.cprogramming.com/reference/preprocessor/token-pasting-operator.html

You call the macro with parameters and the macro is compiled by the preprocessor. The macro helps to generate quellcode for the menu structure.

DmitryGoldobin, 2019-05-16:
Thanks for the reply. Of course, I understand that it is a macro, but unfortunately I can't find the place of definition for "function". I was hoping they'd tell me that, not make a General comment. 

And if it is not difficult, is it possible to view the results of the preprocessor. That is, how to make a command line for a complex project (many files .ino and .h) in the Arduino IDE. Perhaps there is a place with temporary files or special settings that will make it very simple.

I would be very grateful for a specific answer. I reread the C++ specification according to the 2016 standard.